jaromaz/MacintoshPi #2: Does this work on the standard Pi Zero W?

RealAquilus, 2022-07-22:
The readme tells that this doesn't work on the Raspberry Pi 4 but no information are given about Raspberry Pi Zero W

jaromaz, 2022-07-22:
I don't own a Raspberry Pi Zero W and just haven't tested it.
Let us know if it works.

aptonline, 2022-07-31:
I can confirm it works great on a Pi Zero W2 although Mac OS 9 is a tad sluggish so will be even slower on the Zero W.

closed 2022-09-02 by jaromaz

mwy912, 2022-09-23:
I am trying on a Pi Zero W and the screen scrolls horizontally.... Rather than being a static picture, it moves across the screen from left to right.  I'm not sure if that's an issue with the pi zero not supporting, or I need to do something with my setup.

https://user-images.githubusercontent.com/66441897/191960448-bc728ab1-bb49-4231-a425-fd04b04b37cd.mov