jaromaz/MacintoshPi #6: Location of .img or .dsk files?

aptonline, 2022-08-01:
I’m referring to the following in the read me.

The additional file `.img` or `.dsk` can be added to the launched system by adding the image file name at the end of the command which invokes the system concerned, e.g.

`mac os9 demo.img`

My question is where should these files be kept? Can they be added to `Downloads` or do they need to be in the Mac OS path in question? (`/usr/share/macintoshpi/macos9` for example)

aptonline, 2022-08-01:
I think I’ve answered this now, you have to be in the same directory as the `.img` or `.dsk` you want to mount when running the mac os7 command.

jaromaz, 2022-08-01:
You can also give the full path to the image file as well.

closed 2022-08-01 by aptonline