jaromaz/MacintoshPi #14: HD size within Basilisk II / 7.5.5

cannyboy, 2023-02-06:
I'm thinking of creating a "Reference Machine" on a MacintoshPi-ed Pi 3B. Which has lots of old late-90s reference works like Encarta, World Book, Groliers Encyclopaedia, atlases, and various large educational CD-ROM titles.

To do so, I'd like to put them all on the Pi. Either installed or as mountable images. 

But as far as I know, 7.5.5 only allows a maximum of 2GB per HD. So how would I get all the stuff readily available on the MacintoshPi? On a 32GB SD card? Would I have to create new disks somehow?

Latchkey1998, 2023-02-07:
7.5.5 can view a hard drive that is any size but cannot boot from a disk larger that 4 or 3.5 gigs

cannyboy, 2023-02-08:
Update:

There's probably a lot of ways to do this. I ended up using 'Basilisk II GUI' (macOS) to create a 1GB `volume`. 

https://www.emaculation.com/forum/viewtopic.php?t=10454

Then I copied the created `volume` to the SD card's `/usr/share/macintoshpi/macos7` folder using `scp`.

I added this line near the top of `/usr/share/macintoshpi/macos7/macos7.cfg`

`disk   /usr/share/macintoshpi/macos7/1GBHD.dsk`

When you start System 7 on the MacintoshPi again, it will ask you to format it. So that's how you add a disk. 

But I found out that the `macos7.cfg` can mount most disk images (.cdr, .iso, .toast) as either `disk` or `cdrom`. This is useful, because many of these reference apps require the cd to be mounted. So the top of my config file reads something like this:

```
rom    /usr/share/macintoshpi/macos7/Performa-630.ROM
disk   /usr/share/macintoshpi/macos7/hdd.dsk
disk   /usr/share/macintoshpi/macos7/games.dsk
disk   /usr/share/macintoshpi/macos7/1GBHD.dsk

cdrom  /usr/share/macintoshpi/macos7/Encarta95.toast
cdrom  /usr/share/macintoshpi/macos7/EyewitnessScience.toast
cdrom  /usr/share/macintoshpi/macos7/NewWayThingsWork.toast
cdrom  /usr/share/macintoshpi/macos7/TheWayThingsWork.iso
```

closed 2023-02-26 by jaromaz