jaromaz/MacintoshPi #26: Error - can't download: Mac OS 9 ROM file

hammer321, 2023-08-25:
This seems to be the issue: The certificate's owner does not match hostname ‘www.redundantrobot.com’.  I get a certificate mismatch when trying to load that site in Safari as well.

From the installer:
------------------------------------
--2023-08-25 10:41:49--  https://www.redundantrobot.com/sheepshaver_files/roms/newworld86.rom.zip
Resolving www.redundantrobot.com (www.redundantrobot.com)... 66.135.30.216, 2001:19f0:1000:1a47:5400:4ff:fe7c:1414
Connecting to www.redundantrobot.com (www.redundantrobot.com)|66.135.30.216|:443... connected.
The certificate's owner does not match hostname ‘www.redundantrobot.com’

***********

Error - can't download: Mac OS 9 ROM file
Check your Internet connection and try again later.

If you still feel its a bug, then please create an issue here:
https://github.com/jaromaz/MacintoshPi/issues/new

Terminated
------------------------------------

Thank you -Sean

glowrocks, 2023-08-27:
I ran into this same issue. What I did was to download the files manually (see source for which files) by telling Safari to ignore the cert warnings.

I then commented out the line that downloaded and unzipped the file.

It works fine after that.

jaromaz, 2023-09-19:
No certificate check during download has been added to the main branch. It is now working properly.

closed 2023-09-19 by jaromaz