jaromaz/MacintoshPi #27: Keyboard/Mouse Not Working (7 & 8; works for os9)

glowrocks, 2023-08-27:
The subject defines the issue. I've tried two separate installs to try and eliminate user errors. Not sure where to start. The log file has a warning that exists in the 9 log as well, so I don't think that's it.

7 & 8 work though. When I reboot after power cycling, they complain about not rebooting properly. And while running the clock updates regularly and otherwise it seems to be working fine ...other than no keyboard and no mouse.

Oh, I've tried multiple keyboards, including an apple keyboard. Wired and wireless. None work.

Thaanks!!

jaromaz, 2023-09-19:
A very strange case, I have never encountered it. Has the problem already been solved?

jaromaz, 2023-11-08:
Try now - before that read the new [installation section](https://github.com/jaromaz/MacintoshPi#installation).

closed 2023-11-13 by jaromaz