jaromaz/MacintoshPi #29: Terminal blanks out then returns, nevers starts

Logic-Beach, 2023-10-06:
I did as instructed, installed a fresh Raspberry Pi OS (legacy) image onto a RPI 3 B
Everything successfully installed, but when I try to run `mac os7`  in a terminal OR a ttyl console All I get is a briefly blanked out terminal/console then it returns, no errors or output whatsoever.

Am I doing something obviously wrong?

Thanks in advance, it looks like a ton of work went into this.

Logic-Beach, 2023-10-06:
I was able to get logs from `syslog`

Here is the raw output while the terminal is blank after running `mac os7` 

```
Oct 05 21:35:04 raspberrypi systemd[615]: Starting Sound Service...
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Successfully made thread 1940 of process 1940 (n/a) owned by '1000' high priority at nice level -11.
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Supervising 2 threads of 2 processes of 1 users.
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Supervising 2 threads of 2 processes of 1 users.
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Successfully made thread 1943 of process 1940 (n/a) owned by '1000' RT at priority 5.
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Supervising 3 threads of 2 processes of 1 users.
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Supervising 3 threads of 2 processes of 1 users.
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Successfully made thread 1944 of process 1940 (n/a) owned by '1000' RT at priority 5.
Oct 05 21:35:04 raspberrypi rtkit-daemon[837]: Supervising 4 threads of 2 processes of 1 users.
Oct 05 21:35:04 raspberrypi systemd[615]: Started Sound Service.
Oct 05 21:35:04 raspberrypi bluetoothd[594]: Endpoint registered: sender=:1.74 path=/MediaEndpoint/A2DPSource
Oct 05 21:35:04 raspberrypi bluetoothd[594]: Endpoint registered: sender=:1.74 path=/MediaEndpoint/A2DPSink
Oct 05 21:35:04 raspberrypi bluetoothd[594]: RFCOMM server failed for Headset Voice gateway: rfcomm_bind: Address already in use (98)
Oct 05 21:35:04 raspberrypi bluetoothd[594]: RFCOMM server failed for :1.74/Profile/HSPHSProfile/00001108-0000-1000-8000-00805f9b34fb: rfcomm_bind: Address already in use (98)
Oct 05 21:35:04 raspberrypi pulseaudio[1940]: E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
Oct 05 21:35:06 raspberrypi sudo[1945]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/killall pulseaudio
Oct 05 21:35:06 raspberrypi sudo[1945]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 05 21:35:06 raspberrypi sudo[1945]: pam_unix(sudo:session): session closed for user root
Oct 05 21:35:06 raspberrypi bluetoothd[594]: Endpoint unregistered: sender=:1.74 path=/MediaEndpoint/A2DPSource
Oct 05 21:35:06 raspberrypi bluetoothd[594]: Endpoint unregistered: sender=:1.74 path=/MediaEndpoint/A2DPSink
Oct 05 21:35:06 raspberrypi bluetoothd[594]: Endpoint unregistered: sender=:1.70 path=/MediaEndpoint/A2DPSource
Oct 05 21:35:06 raspberrypi bluetoothd[594]: Endpoint unregistered: sender=:1.70 path=/MediaEndpoint/A2DPSink
Oct 05 21:35:06 raspberrypi systemd[615]: pulseaudio.service: Succeeded.
Oct 05 21:35:07 raspberrypi sudo[1948]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/pulseaudio -D
Oct 05 21:35:07 raspberrypi sudo[1948]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 05 21:35:08 raspberrypi bluetoothd[594]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSource
Oct 05 21:35:08 raspberrypi bluetoothd[594]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSink
Oct 05 21:35:08 raspberrypi sudo[1948]: pam_unix(sudo:session): session closed for user root
Oct 05 21:35:08 raspberrypi pulseaudio[1951]: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
Oct 05 21:35:09 raspberrypi sudo[1957]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/amixer -q sset Master playback 40%
Oct 05 21:35:09 raspberrypi sudo[1957]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 05 21:35:09 raspberrypi pulseaudio[1951]: ALSA woke us up to write new data to the device, but there was actually nothing to write.
Oct 05 21:35:09 raspberrypi pulseaudio[1951]: Most likely this is a bug in the ALSA driver 'snd_bcm2835'. Please report this issue to the ALSA developers.
Oct 05 21:35:09 raspberrypi pulseaudio[1951]: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
Oct 05 21:35:09 raspberrypi sudo[1957]: pam_unix(sudo:session): session closed for user root
Oct 05 21:35:09 raspberrypi sudo[1960]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/aplay -q /usr/share/macintoshpi/sounds/os7.wav
Oct 05 21:35:09 raspberrypi sudo[1960]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 05 21:35:10 raspberrypi sudo[1960]: pam_unix(sudo:session): session closed for user root
Oct 05 21:35:10 raspberrypi sudo[1963]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/amixer -q sset Master playback 80%
Oct 05 21:35:10 raspberrypi sudo[1963]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 05 21:35:10 raspberrypi sudo[1963]: pam_unix(sudo:session): session closed for user root
Oct 05 21:35:11 raspberrypi sudo[1967]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/killall pulseaudio
Oct 05 21:35:11 raspberrypi sudo[1967]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 05 21:35:11 raspberrypi sudo[1967]: pam_unix(sudo:session): session closed for user root
Oct 05 21:35:11 raspberrypi bluetoothd[594]: Endpoint unregistered: sender=:1.75 path=/MediaEndpoint/A2DPSource
[... cut ...]
Oct 05 21:35:12 raspberrypi bluetoothd[594]: Endpoint registered: sender=:1.76 path=/MediaEndpoint/A2DPSource
Oct 05 21:35:12 raspberrypi bluetoothd[594]: Endpoint registered: sender=:1.76 path=/MediaEndpoint/A2DPSink
Oct 05 21:35:12 raspberrypi sudo[1971]: pam_unix(sudo:session): session closed for user root
Oct 05 21:35:12 raspberrypi pulseaudio[1974]: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
```

amiblau, 2023-10-21:
I have a very similar problem. In os7,8 I get the startup sound then blank screen and return to command prompt. In os 9 sane but no startup sound.  I am using pi0 2w with legacy os. Amazing work! I am really looking forward to getting this working.

jaromaz, 2023-11-08:
Try now - before that Read the new [installation section](https://github.com/jaromaz/MacintoshPi#installation).

closed 2023-11-13 by jaromaz

gizmochapman, 2024-11-14:
I'm running into this same issue. I have attempted to installation multiple times with Version 1.4.1 without any resolution to this. How was this issue resolved?