jaromaz/MacintoshPi #30: SDL2 not being built?

rainbowmst3000, 2023-10-20:
I am using a Zero 2 W with Buster 32bit and following the instructions exactly. However, the compliation of SDL2 appears not to be triggered and there is no indication that it is happening resulting in Basilisk not being compiled. Any insights?

[image]

RobRS7, 2023-10-27:
I too am receiving SDL2 errors when trying to install MacintoshPi on a Pi Zero 2 W running under 32bit Pi OS (Legacy)
I get an SDL2 error during the installation of Basilisk II -

[image]

Another SDL2 error during the installation of SheepShaver -

[image]

HTH?

brnrds-git, 2023-11-02:
+1, on a Raspberry Pi 3 running a "full" legacy install and following all instructions.

jaromaz, 2023-11-05:
buster is no longer Legacy, it is now the newer bullseye distribution. 
https://www.cnx-software.com/2023/10/12/raspberry-pi-os-upgraded-to-debian-12-bookworm/

For now, I suggest you try the build on the image available below (haven't tried it yet):
https://downloads.raspberrypi.org/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2022-04-07/

Since all this software is legacy, I will try to build MacintoshPi images soon and make them available for download here.

rainbowmst3000, 2023-11-06:
Thanks, but this doesn't work either. Same issue. Also, I tried both Buster and Bullseye. Neither work.

jaromaz, 2023-11-06:
I'm going to check it out next week.

jaromaz, 2023-11-08:
All fixed - read the new [Installation section](https://github.com/jaromaz/MacintoshPi#installation).

rainbowmst3000, 2023-11-09:
Thanks so much!

closed 2023-11-09 by rainbowmst3000