jaromaz/MacintoshPi #31: Can't Download Mac OS 9 ROM File

rainbowmst3000, 2023-11-10:
[image]
Hi Sorry for bothering you again but there seems to be an issue with OS 9. The other two work fine, but OS 9 seems to have a problem. I have tried this at a few different times in case the server was down but no luck. Any suggestions? Thanks.

jaromaz, 2023-11-11:
For some reason the ROM file is not downloading at your location: https://smb4.s3.us-west-2.amazonaws.com/sheepshaver/apple_roms/newworld86.rom.zip

I checked a while ago and it is downloading correctly.

rainbowmst3000, 2023-11-12:
[image]
The files in the macos9 directory under usr/share/macintoshpi/macos9 were set to have a user of "root." I renamed this directory to macos9old and created a new directory for macos9 and copied over the macos9.cfg file. Then everything worked. Thanks! It's great.

closed 2023-11-12 by rainbowmst3000