jaromaz/MacintoshPi #39: -bash: mac: command not found

gizmochapman, 2024-11-13:
I have attempted this project quite a few times with a Zero 2W and a Pi 3B, using both Buster and Bullseye raspios files, each a few times to no avail. Each time I have attempted to use the "mac os7" "mac os8" and "mac os9" commands, I am met with '-bash: mac: command not found'. I cannot seem to navigate around this issue. Please advise if possible!

evitiello, 2024-12-10:
I'm having a similar (maybe the same) issue.  Once installed, if I run 'mac os7' or any of the options, the screen goes black for a moment and I'm returned back to the command prompt.

AxeNinja, 2025-01-22:
I have the same issue. I have tried reinstalling several times, but each time the same result persists; after the install, nothing happens when trying to run "mac os7" or any of the other OS emulations. I have to assume this project is broken at the moment

AleeR78, 2025-02-21:
Chiming in to say it's Feb 21, and I am also having this issue. Installation got hung up a couple of times, but finally managed to work as far as I could tell.

But like the other here, once I'm in rasbian and use the "mac os7" or similar command, it goes black for a few seconds then returns to the default terminal prompt. Any way to fix this?

jaromaz, 2025-09-06:
Since the package repositories required to build MacintoshPi are no longer functional, I’ve made an SD card legacy image of MacintoshPi available for download on my private project page:
https://jm.iq.pl/macintoshpi-mac-os-7-8-9-for-raspberry-pi/