jaromaz/MacintoshPi #43: Mouse lag on 7/8/9

gibbsjoh, 2025-09-06:
Mouse is laggy on all versions - BasiliskII and SheepShaver.

Raspberry Pi 0W

I've tried adding "usbhid.mousepoll=0" and "usbhid.mousepoll=8" to /boot/cmdline.txt (per guides online), no change. Also tried sdl_vsync true and false - again no change. It's not a MASSIVE lag, it's just enough to be annoying!

I recall on Windows with BII, setting Refresh Rate to Dynamic fixes this but I am unable to find the option for the prefs file.

This is my MacOS 7 config file:

`rom    /usr/share/macintoshpi/macos7/Performa-630.ROM
disk   /usr/share/macintoshpi/macos7/hdd.dsk
disk /home/pi/mac/Applications
frameskip 2
ramsize 67108864
ether slirp
modelid 5
cpu 4
fpu true
nosound false
nocdrom false
nogui false
jit false
jitfpu false
jitdebug false
mousewheelmode 1
mousewheellines 3
mixer
ignoresegv true
idlewait true
seriala /dev/tnt1
serialb /dev/null
udptunnel false
extfs /home/pi
screen win/800/600`

jaromaz, 2025-09-06:
Mouse lag is smallest in Mac OS 7 when first selecting the lowest resolution for Raspberry Pi and then the lowest for Mac OS 7. Help is available by entering the command: mac --help
It’s worth checking my sample configurations (available somewhere deeper in this git repo) to easily create a parameter for the mac command that changes both the monitor resolution and the resolution in Mac OS.

Since the package repositories required to build MacintoshPi are no longer functional, I’ve made an SD card image of MacintoshPi available for download on my private project page:
https://jm.iq.pl/macintoshpi-mac-os-7-8-9-for-raspberry-pi/

gibbsjoh, 2025-09-06:
Ah thanks, I'll give that a go!